olkal/HX711_ADC #91: Reading data from Teensy4.1

ICYou, 2022-01-15:
Hi,

I'm working on a controller for my espresso machine. Measuring weight, temperature, controlling pumps and a boiler.
Initially I started the project on Arduino, and have since moved to a Teensy4.1.

Moving to Teensy came with some challenges. One of which was to read the two hx711's. Initially I had output going to -50000 and similar calibration factors. I figured out that the Teensy processor is too fast and the library allows for a delay to be set on the (documented to be more specific to esp32, issue #35 ). So I got the teensy working after setting that delay.

So, I tinkered further, adding code to read out the max31865 for temperature.
And suddenly both hx711's start outputting 0.0.

So, I check if I made any changes that would disrupt reading the hx711s. But I haven't touched any of that code.
So, I continue debugging, writing Serial output statements to get data.
And suddenly I get one working. I continue debugging for the second hx711, and suddenly that one works too.
So I delete the debugging Serial.println statements, and both of the hx711s are back at 0.0

I'm not sure where to go from here. Could it be another issue related to timing?

I'm also aware that the Bogde library has noted something similar.
https://github.com/bogde/HX711/issues/84

olkal, 2022-01-16:
Hi! Have you tried setting SCK_DISABLE_INTERRUPTS to 1 in config.h?

ICYou, 2022-01-16:
Aha!

Setting SCK_DISABLE_INTERRUPTS did not do much.
However, your second comment did the fix. Reading the temperature every 50ms instead made it work. I also see that other people are having a problem with that 75ms delay that they have in the `readRTD` function.

I suppose this problem came partly from my inexperience. A better fix would be to edit the adafruit max31865 library to remove the delays.

Would you happen to have an idea why adding the serial.println lines would enable first one, then both the hx711s to be read? Did it alter the looptiming enough so that both could be read?

olkal, 2022-01-17:
Great! 

A library with blocking functions like the one you are using from Adafruit is very inefficient. Even if you only call it every 50ms like you do, your Teensy will still spend most of it's time doing nothing but wait for the delays to end. It will work okay if all you want to do is to read the temperature, but combined with other tasks it soon becomes a problem. 

On why your code worked better with serial.print, I guess the same as you, by changing the loop timing you just got enough headroom for the HX711 readout to work.

closed 2022-01-30 by olkal