olkal/HX711_ADC #94: Slow Reading Updates

ajkochev, 2022-02-17:
I'm new to Arduino coding so I might be doing something wrong.  I've used a lot of your example code and read over your documentation but cannot find an answer.

My code works, but it can take upwards of 5 or 7 seconds after the weight is put on the scale before I get an accurate reading.  I can live with a 1 or 2 second delay for an accurate reading but 5 to 7 is too much.  I've tried upping the config.h samples line to 32.  I'm also using the WeightScale.update() all over my code as well.  Still cannot seem to get the reading to be accurate any faster.

I am using a cheap HX711 board and have a Sparkfun one on the way.  Hoping this helps as well.  Any help is appreciated.

ajkochev, 2022-02-17:
`
// Version 0.5-2-2022  (Version)-(Month)-(Year) Released
// Created by Anthony J. Kochevar for N Scale Working Railcar Weight Station on Thingiverse.com
// This code is public domain and can be copied, modified and freely distributed.

#include <Wire.h>
#include <HX711_ADC.h>  // Library for HX711 Sensor and Board: https://github.com/olkal/HX711_ADC
#include <LiquidCrystal_I2C.h>  // Library used:  https://github.com/johnrickman/LiquidCrystal_I2C

// Define I2C Address - change if reqiuired.  Default is usually 0x27.
LiquidCrystal_I2C lcd(0x27,20,4);  // Initalize I2C Library for LCD with 16x2(use 20,4) screen.

// Set Button Inputs
int TareCalButton = 6;
int WeightButton = 7;
int SettingsButton = 8;

boolean initCal = false;
int timeDelay = 3000;  //Set LDC display time per line.  3000 = 3 seconds. Use only 1500, 3000 and 5000 as values.  See Settings button in loop function.

unsigned long startTime = 0;
unsigned long startButton = 0;
unsigned long endButton = 0;

float grams = 0;

boolean secondLine = false;  //  User to set if second line should be displayed
boolean backlightOn = true;

//pins:
const int HX711_dout = 4; //mcu > HX711 dout pin
const int HX711_sck = 5; //mcu > HX711 sck pin

//HX711 constructor:
HX711_ADC WeightScale(HX711_dout, HX711_sck);  // Init of library.  "WeightScale" will be the name of the scale.
  

// F() used in serial and LCD output as strings do not need saved in RAM.

  void setup()
  {
    Serial.begin(57600);
    lcd.begin(16,2);
    lcd.backlight();
    delay(2000);
    Serial.println(F("Model Railroad Railcar Weight Scale by Anthony J. Kochevar."));
    Serial.println(F("Version 0.5-2-2022"));  // Change Version to match top comments.
    lcd.clear();
    lcd.setCursor(0,0);
    lcd.print(F(" Model Railroad "));
    lcd.setCursor(0,1);
    lcd.print(F("  Weight Scale  "));
    delay(5000);
    lcd.clear();
    lcd.setCursor(0,0);
    lcd.print(F(" By AJKochevar"));
    lcd.setCursor(0,1);
    lcd.print(F("ver 0.5-2-2022"));  // Change Version to match top comments.
    delay(5000);
    lcd.clear();
    lcd.setCursor(0,0);
    lcd.print(F("Starting up..."));
    Serial.println(F("Starting up..."));
    
    pinMode(TareCalButton, INPUT_PULLUP);  // Tare and Callibration Button
    pinMode(WeightButton, INPUT_PULLUP);  //  Weight Button
    pinMode(SettingsButton, INPUT_PULLUP);  // Settings and Cancel Button

    delay(2000);  // Power up stabilizing time of 2 seconds.
    WeightScale.begin();
[... 395 lines cut ...]
            lcd.setCursor(0,1);
            lcd.print(F("5 seconds  "));
            delay(timeDelay);
            loop();
          }
          if (timeDelay == 5000) {
            SettingsButtonPress = false;
            timeDelay = 1500;
            Serial.println(F("1.5 seconds  "));
            lcd.setCursor(0,1);
            lcd.print(F("1.5 seconds  "));
            delay(timeDelay);
            loop();
          }
        }
        if (endButton >= startButton + 5000) {  // Possible future routine for settings button
          SettingsButtonPress = false;
        }

      }
      WeightScale.update();  //  Always update weight data in while loop to keep data current
      endButton = millis();  //  Always set endButton time in the main while loop for button beening pressed and held.
    }
  }
  `

olkal, 2022-02-18:
Hi!
Ordering a quality board from Sparkfun is a good idea, but I doubt it will be any faster.
I'l try to help you. 

First of all: **never** call `loop()` from anywhere in your code. You are creating a new loop inside itself over and over again (recursion) eating up all available RAM. If you vant to exit from a function you should call `return`.

There is a lot of code here but I have only looked at the function `get_weight()`:
- You probably don't need the 1000ms delay, try to cut it down to e.g. 200ms.
- Your for-loop is looping 20 times, thats not enough to refresh the data set if it is set to 32 samples. Better replace the whole for-loop with the library function `refreshDataSet()`.
- Provided that you have set your HX711 to low rate 10SPS (as you should), 32 conversion takes about 3.2 seconds. Normally 16 conversions is enough, so I would change config.h back to default.  
- It should now take <2 seconds from you push the button until you see det value.

It's possible to speed up the HX711 itself by connecting an external 20mhz crystal, but I don't think you will need to do that.

Also, you use delay a lot in your code. Some places it might be okay, especially for testing, but it can make your code sluggish. If you need something to happen in a sequence with a delay in between you may consider using some kind of task scheduler. There are libraries for this but you can also make it yourself, see the simple example below.

```
//example code

int taskCounter = 0;
long taskTimer = 0;

void taskMachine() {
  if (taskTimer < millis()) {
    switch (taskCounter) {
      case 1:
        task1();
        taskTimer = 2000 + millis();
        taskCounter++;
        break;
      case 2:
        task2();
        taskTimer = 3000 + millis();
        taskCounter++;
        break;
      case 3:
        task3();
        taskCounter = 0;
        break;
    }
  }
}

void task1() {
  Serial.println("This is task 1. Wait 2 seconds, then do task 2");
}

void task2() {
  Serial.println("This is task 2. Wait 3 seconds, then do task 3");
}

void task3() {
  Serial.println("This is task 3. Done.");
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  taskCounter = 1;
}

void loop() {
  // check if there are any tasks to do
  if (taskCounter) {
    taskMachine();
  }
  
  // do something else in the loop while waiting
  static long next_update = 0;
  if ((millis() > next_update) && taskCounter && (taskTimer > 0)) {
    Serial.println(taskTimer - millis());
    next_update = millis() + 250;
  }
}
```
Result:
```
This is task 1. Wait 2 seconds, then do task 2
2000
1749
1498
1247
996
745
494
243
This is task 2. Wait 3 seconds, then do task 3
2993
2742
2491
2240
1989
1738
1487
1236
985
734
483
232
This is task 3. Done.
```

ajkochev, 2022-02-19:
Thank you for the help.  Changing to the `refreshDataSet()` helped a lot.  I'm new to Arduino.  I'm hoping you can help me further to get this project working.  I would be really greatful.  I really like this library and it's the only one I've been able to get working with this project.

As for calling` loop()` I tried` return` but it broke what is being displayed on the LCD at times.  I need, once a function ends, to return to the start of loop to get the screen setup correctly.  Could I move everything in loop to its own function and just have loop call this function to prevent the memory issue?

I am using `millis()` in my code at times to wait while doing other things.  My project is likely to be used mostly by older gentlemen, text needs to pause at times and not proceed so they know what to do at times.  I built in an adjustable setting so they can speed up the delays or slow them down more.

I got my code working on the cheap HX711 board with your help.  I tried the Sparkfun board, but on boot up I get the "Sample to high" error in my code.  I removed the rate connection on the back of the board, I think I set it to open.  No error on bootup but weight readings are not accurate.  I think I'm not understanding "samples" in the project and how to get them set correctly.

olkal, 2022-02-20:
Hi!
When you exit from a function called from the loop you will continue in the loop where you left, you can not jump and start on top. But this is easy to solve; just put the code for correcting the LCD etc into a function and call that function before you return to the loop.
Like this:
```
void myFunction() {
  //do something...
  //write to the LCD etc...
  doBeforeReturningToLoop();
}

void doBeforeReturningToLoop() {
  lcd.clear();
  Serial.println(F("Scale Ready... place rail car on track and press button."));
  //and so on...
}

void loop() {
  if (someCondition) {
    myFunction();
  }
}
```

The sparkfun board shall have the rate jumper closed for 10SPS rate, you should close it again. If you run the test.ino sketch you should get "HX711 measured sampling rate HZ:" value of 10-11. If not I think there is something else going on, double check your wiring.

ajkochev, 2022-02-20:
Thank you oklal for all your help on this project!  I got everything working really well now.  
Making the changes to the sketch to use returns and not call loop seemed to get everything stable.  I got my Sparkfun board to work too.  I kept the jumper open, so I think I'm operating at 80 sps.

I added the following to my code just before reading the data:

`while (WeightScale.update())  {};  // Update scale weight data until smooth.
 WeightScale.refreshDataSet();`

Reading weight now only takes a partial second and is always perfectly accurate.  This wasn't the case sometimes.

Thanks again for all the help to an Arduino noob.

olkal, 2022-02-21:
That's great!

Just to explain the trade-off between speed and accuracy:

HX711 hardware setting:
The HX711 can operate at LOW rate 10SPS (samples/second) or HIGH rate 80SPS. LOW rate is more accurate but obviously slower than HIGH rate. The HIGH rate will increase the microcontroller's workload as it will need to read the conversion values from the HX711 8 times more often to keep up.

Library file config.h setting:
`#define SAMPLES` defines the size of the library data set used for calculating the rolling average output value. A high value is more accurate but slower. A high value also takes up more dynamic memory from the microcontroller.

Unless you need very fast settling time you will usually have best results by setting the HX711 rate setting to LOW/10SPS, and then adjust `#define SAMPLES` value in config.h to achieve an acceptable balance between speed and accuracy. This way you are basically moving a lot of the workload from the microcontroller to the HX711.

But if you now have it working and are happy with both accuracy and speed, perhaps it's better to keep it as it is.

olkal, 2022-02-22:
Issue considered solved.

closed 2022-02-22 by olkal