olkal/HX711_ADC #95: Unstable value

wrinkledmind, 2022-02-27:
After calibrating, I uploaded the Read 1x load cell, the value keep changing, drifting or unstable.

The load is located very near my breadboard and I don't think there is a temperature difference.
The load cell is at rest.

even if I call the tare function, the value keep drifting.
It could drift up to 30 grams.
Even when I put weight, the value unstable.

[https://imgur.com/a/ErJTRHP](https://imgur.com/a/ErJTRHP)

olkal, 2022-02-27:
Are you powering the HX711 from 3.3V? If so, try to set it up with a 5v board and see if it improves (the HX711 internal voltage regulator will not work properly at 3.3v supply). Could also be the module itself, some of the chinese modules are not designed well.

wrinkledmind, 2022-02-27:
Thanks for your fast reply.
I added to the link above, a graph showing values of the raw data over the course of 2 hours.
I am powering from 5V.
The voltage across VCC and GND is 5V while the voltage across E+ and E- is 4.15V
Vbg=1.27V ( voltage across pin6 and 5)
Hope these information help you trace the problem.

The values not close to my 10$ kitchen scale.

olkal, 2022-02-27:
I can't say for sure that the board design is the (single) cause here, but the module in the picture is one of the modules that differs most from the reference design.
- E- and pin 5 should be connected to GND and it's not
- The line from A- to pin 7 is missing the serial resistor, so the input filter for A-/A+ is not working properly
- Capacitor C7 is not located close to transistor Q2 as it should be
- The voltage across E+ and E- is expected to be more like 4.35V, this is not a problem in itself, but I would guess that the reason for it being 4.15V is that E- is not grounded.

closed 2022-04-13 by github-actions[bot]