olkal/HX711_ADC #96: Regarding compatibility with Arduino Stm32 official core

anandspathak, 2022-03-03:
THIS IS NOT AN ISSUE, Thanks for this great library. I can use with arduino uno,
Is this library works with Arduino Stm32 official core ( LINK:  https://github.com/stm32duino/Arduino_Core_STM32)  ??
By the way library complies without any error with above stm32 core.
Thanks,
Regards,
ANAND PATHAK.

olkal, 2022-03-04:
Hi!
I have not tested the library with STM32 core, but if it compiles I don't see any reasons that it should not work.

anandspathak, 2022-03-04:
Thanks, I will try and revert with result.

anandspathak, 2022-04-08:
Sorry for being late...I finished my project on arduino uno...not tried on stm32 as of now...

anandspathak, 2022-04-08:
Sir, 
This is 50 tons load cell and I have no idea how to calibrate..Actually this load cell is attached to when piston  moves down due to hydraulic pressure , it presses the load beneath it and then goes up...and again comes down presses and goes up..this cycle repeats 2 times in one second....
My issue: so far I have only used bar load cell max 20 kgs which I can calibrate easily as the test mass during calibrating is kept on top of load cell, but in this case I don't figure out how can I calibrate...

What I tried: I applied hydraulic pressure so that load cell presses against load...the pressure was of 20 Tons... So in calibration sketch I entered 20000000 grams..mine calibration factor was -0.00...where I was wrong??
pls suggest something...
Thanks and Regards
Anand Pathak,
INDIA

olkal, 2022-04-08:
Hi!
Thats a big load cell and I don't think the library will work with gram as units, could you try using 1kg as units (20000) or 10kg  (2000)?

anandspathak, 2022-04-09:
Thanks for your quick reply,
As of now the load cell isn't with me....so I cant check...will see later....
Thanks

anandspathak, 2022-09-14:
> Hi! Thats a big load cell and I don't think the library will work with gram as units, could you try using 1kg as units (20000) or 10kg (2000)?

Exactly I did as you suggested....
I got chance to calibrate a 10 tons load cell..actual weight was 1.00 tons ....what i did was entered a random value in calibration...noted down the displayed weight value which was much more than 1 ton...again entered a random calibration value more than previous one...found that the displayed weight was reducing but still was more than actual 1 ton load....so i again entered calibration value more than previous one...this time the actual weight and displayed weight were very close to each other...few more fine adjustments in calibration value resulted in actual value as 1 ton....
And moreover in actual test by increasing and decreasing the test load the weight also changes correctly... I mean to an extent of precision of 2 decimal digits...and stable in longer continuous runs (tested continuously for 72 hrs in one stretch)...no random ups and downs....awesome....

olkal, 2022-09-14:
Many thanks for your feedback Anand, and good luck with your project!

anandspathak, 2023-08-07:
Sir,
I have tested your library and confirm it working great on Arduino Stm32 official core:
(LINK: https://github.com/stm32duino/Arduino_Core_STM32)...
My query is that is it possible to read 8 hx711 modules on using mega2560 or stm32 bluepill board??
an example provided shows how to read two hx711 modules, going ahead sameway can i add 6 more hx711 to get readings from each of 8 modules???
please tell me your opinion,
Thanks and Regards
Anand Pathak,
INDIA

olkal, 2023-08-08:
Hello Anand! I don't see any problem by reading 8 pcs hX711, but be sure that you have enough dynamic memory available. 
And yes, you can just use the example for two HX711 and add 6 more instances.

anandspathak, 2023-08-08:
Thanks for your help,
Anand Pathak,
INDIA

closed 2023-08-18 by olkal

anandspathak, 2023-08-19:
Dear Sir,
I am running low on pins for using 8 hx711 modules....can i make  every hx711 clock pins common and use only one pin for clock and individual pins for data????
is it possible???
Thanks for your help,
Anand Pathak,
INDIA

olkal, 2023-08-23:
Hi! I think it's theroretiacally possible to use common clock pin, but it's not supported by this library. To do so you would at least have to read all 4 dout pins for every single clock pulse.