olkal/HX711_ADC #97: using channel A + B together

vina258466, 2022-03-17:
i want to try to use two load cells but only use one hx711 module, the first load cell i connect to channel A and the second load cell i connect to channel B, how can i use this library, please share the solution for my problem

olkal, 2022-03-19:
You can select the B-channel like this: `LoadCell.setGain(32);` and back to A-channel: `LoadCell.setGain(128);`

I don't have the solution or any code example for what you try to acheive. There is only one dataset in the library and until that has been "flushed" with new readings from the selected channel, the average value would be a mix of channel A and B. So to make it work I think you should use two instances of the library, one instance for each channel. Also note that the B-cannel have a gain of only 32, so it's likely to have more noise and less resolution compared to the A-channel.

If possible, use one dedicated HX711 for each loadcell.

vina258466, 2022-03-21:
thank you for the information

closed 2022-05-06 by github-actions[bot]