olkal/HX711_ADC #99: Problem with the loadcell

opapferdi, 2022-04-02:
Hello everyone, I'm trying to build the project. But I have a problem, when weighing, only one cell measures. The other shows a weird value. If I swap the inputs, the other cell will measure and the other will display the weird value. Unfortunately, both together do not work. Can someone give me tips for troubleshooting? I've checked the wiring several times.

moncholin, 2022-04-02:
I am having a similar issue. Testing each load cell separately seems like they are working correctly. The problem is when they are working together. One weighs correctly but the other one shows 0. I am using the sparkfun HX711 drivers.

opapferdi, 2022-04-03:
I have two of this.
https://www.amazon.de/dp/B07LD9N684/ref=cm_sw_r_apan_i_PSKGQ12WVNHSSCHW8S7S

And the front one is always showing - 21147....
Equal if I change the input or not. But after change thr front will measure and represent the value as the rear one.

olkal, 2022-04-03:
Hi! Could it be that you are connecting one of the two HX711 modules to IO pins on the microcontroller that are not suitable ? What kind of mcu har you using?

opapferdi, 2022-04-03:
Hi, i use This mcu
https://www.amazon.de/AZDelivery-PRO-MINI-ATmega328-16MHz/dp/B07VKLCYFV?pd_rd_w=0xP9r&pf_rd_p=6b1afec3-f278-4d3c-84b7-11c87b1dcb9b&pf_rd_r=9SB5F00H79R7Z7WXAB52&pd_rd_r=2052914a-3c86-4ffa-90e8-457143d490fc&pd_rd_wg=NM82p&pd_rd_i=B07VKLCYFV&psc=1&ref_=pd_bap_d_rp_1_t

and the H711 are conected to A0-A3

olkal, 2022-04-03:
Okay, pin A0-A3 on the ATmega328 should be fine. 
Have you tried using the included library example sketch `Read_2x_load_cell.ino` unmodified, except for changing the pin numbers in line 17-20?

opapferdi, 2022-04-03:
Yes i tried it. And the result is like this:
⸮⸮x⸮⸮⸮ꤽ⸮꡼⸮⸮⸮<* ⸮⸮⸮"⸮⸮⸮?* ⸮<*⸮⸮<*#⸮⸮?*⸮⸮<*#⸮⸮>*⸮⸮⸮>*#⸮⸮<⸮⸮⸮⸮?⸮ ⸮<⸮⸮⸮⸮?⸮"⸮⸮<⸮⸮⸮⸮<⸮#⸮⸮<⸮⸮⸮⸮⸮" ⸮?*⸮⸮⸮>⸮ ⸮<*⸮⸮?⸮ ⸮<*⸮⸮⸮>*#⸮⸮?⸮⸮⸮⸮?* ⸮⸮>⸮⸮⸮⸮?⸮"⸮⸮>⸮⸮⸮⸮?⸮ ⸮⸮>⸮⸮⸮<⸮#⸮⸮>⸮⸮⸮⸮?⸮"⸮⸮⸮"⸮⸮⸮?⸮ ⸮⸮>⸮⸮⸮⸮>⸮"⸮⸮?⸮⸮⸮⸮⸮"⸮⸮s⸮#⸮⸮⸮>⸮#⸮⸮⸮"⸮⸮<⸮#⸮⸮>⸮⸮⸮>⸮#⸮⸮⸮"⸮⸮⸮>⸮"⸮⸮?⸮⸮⸮⸮?⸮#⸮⸮>⸮⸮⸮⸮>*#⸮⸮<⸮⸮⸮<*#⸮⸮>⸮⸮⸮⸮?⸮"⸮⸮?⸮⸮⸮⸮?⸮⸮⸮<⸮⸮⸮⸮>⸮⸮⸮8⸮⸮⸮⸮?⸮!⸮?⸮⸮⸮⸮<
!⸮>⸮⸮⸮⸮?*⸮⸮<⸮⸮⸮⸮>.#⸮⸮<⸮⸮⸮⸮?*⸮⸮<⸮⸮⸮⸮>*!⸮s⸮#⸮⸮⸮>*⸮⸮>⸮⸮⸮<*#⸮⸮>⸮⸮⸮<*"⸮⸮?⸮⸮⸮⸮8*#⸮⸮8⸮⸮⸮⸮>* ⸮<⸮⸮⸮⸮>*⸮⸮<⸮⸮⸮⸮>*#⸮⸮?⸮⸮⸮⸮8*⸮⸮s⸮r⸮⸮⸮b⸮⸮<*⸮⸮⸮<⸮⸮⸮?*⸮⸮⸮>⸮!⸮<*⸮⸮⸮>⸮#⸮⸮?*⸮⸮⸮?* 8A(⸮⸮⸮⸮?.⸮⸮s⸮r⸮⸮⸮?*⸮⸮<*⸮⸮⸮⸮r⸮⸮>*⸮⸮⸮r ⸮⸮<*⸮⸮⸮?⸮⸮⸮<*⸮⸮⸮<⸮⸮⸮s⸮r⸮⸮⸮?
 ⸮⸮>⸮⸮⸮⸮⸮c ⸮s⸮"⸮⸮⸮⸮b
⸮⸮<⸮⸮⸮⸮>⸮⸮?⸮⸮⸮⸮>* ⸮>⸮⸮⸮⸮?* ⸮?⸮⸮⸮⸮⸮r⸮⸮>⸮⸮⸮<⸮⸮⸮?⸮⸮⸮⸮<*⸮⸮s⸮⸮⸮<
⸮⸮<⸮⸮⸮⸮>⸮⸮⸮s⸮⸮⸮⸮>⸮ ⸮s⸮⸮⸮⸮?⸮⸮⸮<⸮⸮⸮⸮<
"⸮⸮>⸮⸮⸮<*⸮>⸮⸮⸮<⸮#⸮⸮>⸮⸮⸮⸮⸮b ⸮<⸮⸮⸮⸮?*⸮⸮?⸮⸮⸮⸮<⸮⸮⸮s⸮b⸮⸮⸮>⸮ ⸮⸮8⸮⸮⸮⸮>⸮ ⸮⸮<*⸮⸮⸮룠⸮s⸮b⸮⸮⸮룊⸮>*⸮⸮⸮?⸮⸮⸮>⸮⸮⸮?⸮⸮⸮>
⸮⸮⸮?⸮⸮⸮>⸮⸮⸮<⸮⸮⸮<
⸮⸮⸮?⸮⸮⸮>*⸮⸮⸮<⸮⸮⸮s⸮r⸮⸮⸮⸮"⸮⸮>*$⸮⸮<⸮#⸮⸮>*⸮⸮⸮?⸮"⸮⸮?*⸮⸮>⸮ ⸮<*⸮⸮⸮⸮# ⸮s⸮⸮⸮⸮⸮>⸮⸮⸮s⸮⸮⸮⸮⸮?⸮⸮⸮s⸮r⸮⸮⸮<⸮⸮⸮s⸮s⸮⸮⸮?⸮ ⸮?*"⸮⸮>⸮⸮⸮>*⸮⸮⸮<⸮⸮⸮s⸮r⸮⸮⸮?⸮⸮⸮>*⸮⸮⸮>⸮!⸮<⸮⸮⸮?⸮⸮⸮?⸮⸮⸮⸮?⸮⸮⸮?⸮⸮⸮>⸮⸮⸮>*⸮⸮⸮?⸮⸮⸮>
⸮⸮⸮>⸮ ⸮>⸮⸮⸮⸮?⸮⸮⸮<*⸮⸮⸮?⸮⸮⸮<*<⸮⸮>⸮⸮⸮8*6⸮⸮?⸮⸮⸮>*⸮⸮⸮?⸮⸮⸮>.8⸮⸮>⸮ ⸮s⸮b⸮⸮⸮<⸮	⸮⸮>⸮⸮⸮<⸮
⸮⸮⸮c⸮⸮⸮뢠⸮<⸮⸮⸮⸮<⸮⸮⸮>⸮⸮⸮⸮<⸮	⸮⸮⸮b⸮⸮⸮뢠⸮?⸮⸮⸮⸮?⸮⸮⸮<*⸮⸮⸮?⸮
⸮⸮?*2⸮⸮>⸮⸮⸮<*⸮⸮⸮?⸮⸮⸮8*⸮⸮?⸮
⸮⸮>*⸮⸮⸮8⸮⸮⸮<*⸮⸮⸮>⸮⸮⸮<*⸮⸮⸮룠⸮s⸮r⸮⸮⸮?⸮
⸮⸮>*⸮⸮<⸮⸮⸮⸮s

And much longer. everytime i push on one gauge it start a new line.
Sry i´m new to arduino ;)

olkal, 2022-04-03:
You must set correct baud rate in the Serial Monitor window, select 57600 baud.

opapferdi, 2022-04-03:
Thanks for the tip. So both cells measure!
But if i compile your sketch it schon for one cell the weird value. If i make C1 and C2 or M1 M2 this would be shown:

C1
Start calibration loadcell FRONT:
Place the CG scale an a level stable surface.
Send 'z' from serial monitor to set the zero offset.
Now, place your known mass on the loadcell FRONT, then send the weight of this mass (i.e. 500.0) from the serial monitor.
Invalid value

C2
Known mass is: 2.00
Calculated calibration value is: nan, use this in your project sketch or save to eeprom.
Save this value to EEPROM adress 0? y/n
Value not saved to EEPROM
End calibration
***
***
M1
Current value for loadcell FRONT is:nan
Now, send the new value from serial monitor, i.e. 696.0
Invalid value, exit

Damn i´m not stupid but this hard to understand for me.

olkal, 2022-04-03:
It seems to that you are typing something else than a number when asked to send a value in the Serial Monitor? 
If for example your known mass is 450 gram you should type `450.0` in the Serial Monitor and hit enter. The response `Invalid value` means that the provided value is not more than 0 and the response `Calculated calibration value is: nan` are indicating that the calibration value is "not a number" (nan).

opapferdi, 2022-04-03:
The result is this:
Start calibration loadcell FRONT:
Place the CG scale an a level stable surface.
Send 'z' from serial monitor to set the zero offset.
Now, place your known mass on the loadcell FRONT, then send the weight of this mass (i.e. 500.0) from the serial monitor.
Invalid value
Known mass is: 200.00
Calculated calibration value is: nan, use this in your project sketch or save to eeprom.
Save this value to EEPROM adress 0? y/n
Value nan saved to EEPROM address: 0
End calibration
***
***
Start calibration loadcell REAR:
Place the CG scale an a level stable surface.
Send 'z' from serial monitor to set the zero offset.
Now, place your known mass on the loadcell REAR, then send the weight of this mass (i.e. 500.0) from the serial monitor.
Invalid value
Known mass is: 200.00
Calculated calibration value is: 94.68, use this in your project sketch or save to eeprom.
Save this value to EEPROM adress 4? y/n
Value 94.68 saved to EEPROM address: 4
End calibration
***
***
Start calibration loadcell FRONT:
Place the CG scale an a level stable surface.
Send 'z' from serial monitor to set the zero offset.
Now, place your known mass on the loadcell FRONT, then send the weight of this mass (i.e. 500.0) from the serial monitor.
Invalid value
Known mass is: 200.00
Calculated calibration value is: nan, use this in your project sketch or save to eeprom.
Save this value to EEPROM adress 0? y/n
Value nan saved to EEPROM address: 0
End calibration
***
LdCell_1:-21474836.47   LdCell_2:0.10   Wt:-21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.20   Wt:-21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.16   Wt:-21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.11   Wt:-21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.06   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.21   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.28   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.25   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.15   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:9.05   Wt:-21474827   CG:0.0
LdCell_1:-21474836.47   LdCell_2:47.33   Wt:-21474789   CG:0.0
LdCell_1:-21474836.47   LdCell_2:47.19   Wt:-21474789   CG:0.0
LdCell_1:-21474836.47   LdCell_2:14.58   Wt:-21474821   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.31   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.26   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.11   Wt:21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.32   Wt:-21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.76   Wt:-21474835   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.58   Wt:-21474835   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.68   Wt:-21474835   CG:0.0
LdCell_1:-21474836.47   LdCell_2:1.42   Wt:-21474835   CG:0.0
LdCell_1:-21474836.47   LdCell_2:2.33   Wt:-21474834   CG:0.0
LdCell_1:-21474836.47   LdCell_2:2.66   Wt:-21474833   CG:0.0
LdCell_1:-21474836.47   LdCell_2:2.95   Wt:-21474833   CG:0.0
LdCell_1:-21474836.47   LdCell_2:1.92   Wt:-21474834   CG:0.0
LdCell_1:-21474836.47   LdCell_2:0.36   Wt:-21474836   CG:0.0
LdCell_1:-21474836.47   LdCell_2:-0.89   Wt:21474835   CG:0.0
LdCell_1:-21474836.47   LdCell_2:2.12   Wt:-21474834   CG:0.0
LdCell_1:-21474836.47   LdCell_2:68.07   Wt:-21474768   CG:0.0
LdCell_1:-21474836.47   LdCell_2:143.89   Wt:-21474692   CG:0.0
LdCell_1:-21474836.47   LdCell_2:201.82   Wt:-21474634   CG:0.0
LdCell_1:-21474836.47   LdCell_2:202.10   Wt:-21474634   CG:0.0
LdCell_1:-21474836.47   LdCell_2:202.53   Wt:-21474633   CG:0.0
LdCell_1:-21474836.47   LdCell_2:202.90   Wt:-21474633   CG:0.0
LdCell_1:-21474836.47   LdCell_2:178.37   Wt:-21474658   CG:0.0
LdCell_1:-21474836.47   LdCell_2:102.96   Wt:-21474733   CG:0.0
LdCell_1:-21474836.47   LdCell_2:27.61   Wt:-21474808   CG:0.0
LdCell_1:-21474836.47   LdCell_2:1.91   Wt:-21474834   CG:0.0
LdCell_1:-21474836.47   LdCell_2:1.78   Wt:-21474834   CG:0.0
LdCell_1:-21474836.47   LdCell_2:1.71   Wt:-21474834   CG:0.0

I thank you for your help, but i must go to bed. Tomorow i will test again.

closed 2022-05-19 by github-actions[bot]