olkal/HX711_ADC #100: Can't get the HX711 to read values from load Cell

JayShack, 2022-04-07:
Hi,

I'm using a MEGA 2560 board and set everything up exactly like Indrek did in his youtube tutorial about using a 4 wire load cell and the HX711 board.  Whenever I run the calibration program, the calibration factor always returns as 0.0 and the measurements print as nan after calibration which I assume is because the value it's calculating is 0/0. 

"
Starting...
Startup is complete
***
Start calibration:
Place the load cell an a level stable surface.
Remove any load applied to the load cell.
Send 't' from serial monitor to set the tare offset.
Tare complete
Now, place your known mass on the loadcell.
Then send the weight of this mass (i.e. 100.0) from serial monitor.
Known mass is: 20.00
New calibration value has been set to: 0.00, use this as calibration value (calFactor) in your project sketch.
Save this value to EEPROM adress 0? y/n
Value 0.00 saved to EEPROM address: 0
End calibration
***
To re-calibrate, send 'r' from serial monitor.
For manual edit of the calibration value, send 'c' from serial monitor.
***
Load_cell output val: nan
Load_cell output val: nan
Load_cell output val: nan
Load_cell output val: nan
Load_cell output val: nan
Load_cell output val: nan
Load_cell output val: nan
Load_cell output val: nan
"

 I'm confident in all of my soldered connections and have gotten the load cell to read using the SparkFun OpenScale board which is simpler and has all of the programs loaded onto the board already. I believe my issue is with the code or the way I have it hooked up to the Arduino, which is exactly how it's shown in the Youtube tutorial.

Any ideas or help would be much appreciated.

Thanks,
JayShack

olkal, 2022-04-07:
Hi!
I have had similar reported issues before that have turned out to be incorrect wiring from the HX711 module to the mcu IO pins, or blown or incompatible IO pins. Please double check that the `clk` and `dout` pin/wiring is correct and that the mcu IO pins that you are using are working, and that you have 5V power to the HX711. 
If you don't find anything wrong, could you try the `Read_1x_load_cell.ino ` example sketch and see what kind of output you get with that?

JayShack, 2022-04-07:
Hi,

I just checked the pins using this code while nothing was connected to the board

"
int analog0;
int analog1;
int analog2;
int analog3;
int analog4;
int analog5;
int analog6;
int analog7;
int analog8;
int analog9;
int analog10;
int analog11;
int analog12;
int analog13;
int analog14;
int analog15;

int digi1;
int digi2;
int digi3;
int digi4;
int digi5;
int digi6;
int digi7;
int digi8;
int digi9;
int digi10;
int digi11;
int digi12;
int digi13;
int digi14;
int digi15;
int digi16;
int digi17;
int digi18;
int digi19;
int digi20;
int digi21;
int digi22;
int digi23;
int digi24;
int digi25;
int digi26;
int digi27;
int digi28;
int digi29;
int digi30;
int digi31;
int digi32;
int digi33;
int digi34;
int digi35;
int digi36;
int digi37;
int digi38;
int digi39;
int digi40;
int digi41;
int digi42;
int digi43;
int digi44;
int digi45;
int digi46;
int digi47;
int digi48;
[... 245 lines cut ...]
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
Load_cell output val: 0.00
"

Thanks,
JayShack

JayShack, 2022-04-07:
I couldn't find another way to test the pins, and I'm very new to using Arduino's so I couldn't come up with any ways on my own.

olkal, 2022-04-07:
That's a funny way of testing the IO pins, I'm not sure what the author is trying to achieve but it doesn't tell you anything useful :) Just moving the `dout` and `sck` to some other pins on a different IO port should be sufficient, for example if you use D4 and D5 now, try to move to D6 and D7. Anyway, there doesn't seem to be any communication between your mcu and the HX711. Is the HX711 module you are using from Sparkfun as well, or is it some chinese make? You don't have a spare HX711 module available to check if there is something wrong with the module itself?

JayShack, 2022-04-08:
I've ordered an Arduino Uno and a fresh HX711 module to try again with.  So hopefully I'll have some better luck starting over with fresh components.  I used about half of the pins on the MEGA trying to get everything to work so either they were all bad or I had some other issue.  Thanks for all the help, I'll try again in a few days when the new parts get in.

Thanks

olkal, 2022-04-08:
Okay, good!

JayShack, 2022-04-08:
It worked first time with the new Arduino Uno and fresh HX711 chip, so it must've been some issue between those two.  Thanks for the speedy replies and help!

closed 2022-04-08 by olkal