olkal/HX711_ADC #103: Problem with LoadCell.tareNoDelay() and LoadCell.getTareStatus()

alexandru98765, 2022-05-12:
Hi,

I understand that tareNoDelay() need to complete in order to return getTareStatus() True, but how can i know if tareNoDelay() is completed?

I have trouble modifying this:
if (Serial.available() > 0) {
      if (Serial.available() > 0) {
        char inByte = Serial.read();
        if (inByte == 't') LoadCell.tareNoDelay();
      }
    }

I don't want to enter 't' to serial, i want to push a button in order to tare. I spent so much time trying to implement functions in order to work and it does not. Could you help me with this?

olkal, 2022-05-13:
Hi!
The function `getTareStatus()` will return `true` when the `tareNoDelay()` process is completed.
Example:
```
    if (LoadCell.getTareStatus() == true) {
      Serial.println("Tare complete");
    }
```

To initiate the tare function from a push button input, I suggest that you download the Momentary-switch library https://github.com/olkal/Momentary-Switch and implement it in your sketch. Then call `tareNoDelay()` when button is released.

prashantlsharma, 2022-05-29:
Hello,
I have trouble modifying this:
if (Serial.available() > 0) {
if (Serial.available() > 0) {
char inByte = Serial.read();
if (inByte == 't') LoadCell.tareNoDelay();
}
}

I don't want to enter 't' to serial, since i want to process tareNoDelay() function directly even without giving any command since the whole process will be like when we will begin calibrate function there will be no wait placed on the tray and we want tareNoDelay() perform without even in getting any serial command. I have tried using flag or already pre defined char to perform the function but it never worked.
Please help me with this

closed 2022-07-14 by github-actions[bot]