olkal/HX711_ADC #105: Problem with 2 HX711 and pin occupancy

Jonewbie69, 2022-06-08:
Hello,

first of all many thx for the script and the explanations.

I try to use the board with 2 HX711 and 4 load cells each, to weigh two bottles of gas in my RV (camper). I want to show the result on the onboard display and send them via BT to my tablet.

I have a TTGO-T board with an 1.14 lcd display. For the 1st HX711 I use pin 25 (dout1) and pin 26 (sck1), the 2nd HX I tried nearly every other or combination of pins ....I have changed it on the board and in the sketch. If I use pin32&33 the display shows me "inf" instead of any kilos. If I only use pin 25 & 26 I see the right measure.

Do you have any ideas, what might be a solution ?

Greetings
Jo

olkal, 2022-06-09:
Hi Jo!
I think the GPIO32 and 33 should be okay to use, so I'm not sure. 
Do you have the same problem if you use the library example sketch? 
If you have a LED available you could test the GPIO's with a blink sketch just to be sure that they work.

Jonewbie69, 2022-06-09:
Hello Olav,
thank you for your prompt answer.

I tried your example sketch and it works fine. 

In my modyfied sketch I have combined your sketch with the <TFT_eSPI> , <bluetoothserial> and <SPI>.
The only Probem now is, that the calibrationvalue has not the right factor an 11kg bottle weights 331 kg.
I try further on, but my time is very rare. I hope I can tell you in about 1 month, if I solve the problem or not.

Greets 

Jo
PS: in 1 week I leave to southern norway for holiday.

olkal, 2022-06-09:
Great Jo! The calibration factor should be easy to dial in, I'm looking forward to hear if you manage sort everything out.
Enjoy your stay in Norway!

closed 2022-07-24 by github-actions[bot]