olkal/HX711_ADC #106: tare function does not tare

strud, 2022-06-25:
Hi there

I am calling  the tare() function but getting a large residual reading afterwards of about 9000 to 10000 (cal = 1)

What would be the cause of this?

Do you have a limit on how much offset the tare can cope with?

olkal, 2022-07-16:
Hi, sorry for the delayed response.
There is no limit to the amount of tare offset. I don't know the reason for the issue you have, but perhaps the functions `getTareOffset();` and `setTareOffset();` could be useful to find out what is happening.

dennyakbar, 2022-07-28:
Hi Olkal, 
For the weighing process, can these two functions be used or what? (setTareOffset and setCalFactor), or just one ?

dennyakbar, 2022-07-29:
Hi Olkal, 

function setTareOffset(); and setCalFactor(); can be used both or just one is enough ?

olkal, 2022-08-02:
Hi!
They are totaly different functions and can both be used.
The output is calculated like this (simplified): `outputdata = (data - tareoffset) / calfactor`

dennyakbar, 2022-08-02:
Olkal, thank you.

closed 2022-08-25 by olkal