olkal/HX711_ADC #108: Nordic 52832 Examples and Tut

LiveRock, 2022-07-26:
Hi

I need help understanding how to use the library with the Nordic 52832 with VSC and PlatformIO.
I am using a water bottle of 500 grams as known mass and a Wheatstone Bridge of 4 LoadCells.

I am getting a calibration value of **-12**

float knownMass = **0.5** (kg)

 When I read data, I get reading of 2 Kg when I step on the scale. It should be like 78 Kg.

Questions:
(1) What do I do with **calibrationValue** after I obtain it?
calibrationValue = LoadCell.getNewCalibration(knownMass); 

(2) float i = LoadCell.getData(); 
Is value i in kg?

olkal, 2022-08-02:
Hi!
1) calibrationValue is included like this: `setCalFactor(calibrationValue);` (see example file).
2) The library does not use units, the output value is in whatever unit you used for the known mass when doing calibration. Known mass calibration input in unit kg = output in unit kg.

The calibration value should normally be a positive number, I'm not sure why you get -12. Have you tried to use one of the example files unmodified?
Also, the "known mass" used for calibration should be more in the neighbourhood of the scales capacity, for a scale with capacity of 100kg the "known mass" should be at least 20kg in my opinion. Try using yourself as known mass, you can fine tune this later by comparing with a calibrated scale.

LiveRock, 2022-08-08:
Just wondering, is the issue I am facing related to the fact I am using the Nordic 52832 chip with Arduino framework, Visual Studio Code and PlatformIO?

olkal, 2022-08-09:
No, I don't think so. 
The only thing is that saving the calibration value to EEPROM is not supported for the Nordic chip, so you must put the value in the sketch.

closed 2022-09-23 by github-actions[bot]