olkal/HX711_ADC #111: Load_cell output val: NAN

YPayal, 2022-10-05:
I am getting "Load_cell output val: NAN " my connections are perfectly fine. Can someone help me with this.

olkal, 2022-10-05:
Hi
NAN is short for "not a number" and means that the value is not a valid printable number. 
Similar reported issues have turned out to be incorrect wiring from the HX711 module to the mcu IO pins, or blown or incompatible IO pins. Please double check that the clk and dout pin/wiring is correct and that the mcu IO pins that you are using are working, and that the HX711 have power. If you have another mcu available, try to replace it.

YPayal, 2022-10-06:
[image]
[image]
[image]
I am uploading the photos of my set up, can you let me know where am I going wrong, It would be a great help

olkal, 2022-10-11:
Hi
I can't see anything wrong in the pictures. 
I can't see how you conneted the HX711's VCC and GND, but I assume that they go to the Arduino 5V and GND.
Check that pin 4 and 5 are actually working as outputs by using a LED and the Blink.ino example, or try to use some other output pins.

YPayal, 2022-10-15:
Hey @olkal Thanks for your previous message, the code is working now the issue was with the wiring.

closed 2022-11-15 by olkal

YPayal, 2023-01-16:
Hey @olkal I am doing a need project with arduino, basically I want to set an constant humidity chamber and I am not really good at coding, was hoping if you could help me.

YPayal, 2023-01-16:
Hey @olkal I am doing a new project with arduino, basically I want to set an constant humidity chamber with BME280 sensor and I am not really good at coding, was hoping if you could help me.