olkal/HX711_ADC #112: ESP32 Wroom Pin For CLK

Sudupatil, 2022-10-20:
Hello,

Your library is great, but unfortunately I stuck in one situation, where i would like to have you help in that. In my project I used 3.5 Touch Screen Display(ILI9488), For That i used following pins
T_IRQ	-
T_DO	19
T_DIN	23
T_CS	13
T_CLK	18

SCK	           18
SDI(MOSI)   23
DC	           4
RESET	   2
CS	           15

I was using following pis for HX711_ADC
HX711_dout = 36
HX711_sck = 23

Now due to Pin 23 is in common between Both, I can't Use Pin 23 for HX711, i had tried all other pins  but didn't got any success. 

So my question is Which Other pin i can use for HX711_sck (other than display pins)?

Thanks and Regards
Sudarshan Patil

olkal, 2022-10-20:
Hi!
One of the following GPIO pins should work for HX711 `sck`: 5, 14, 21, 22, 25, 26, 27, 32, 33

Sudupatil, 2022-10-21:
Hello,

Thank you very much for your timely reply, I used Pin 33 as your suggested and it's work.  

Thanks and regards
Sudarshan Patil

closed 2022-10-21 by Sudupatil