olkal/HX711_ADC #113: Question on Smoothing

Abasz, 2022-11-03:
I would like to use your library for a cycle power meter and I have the following questions:

The strain gauges would be placed on the crank arm, the consequences of which will be that there will be negative force during part of the crank revolution (when one pulls the pedal upward).

Is it possible to set the smoothing to use absolute values of the data spit out by the HX711 module? 

Or do you recommend setting the smoothing to 1 (effectively switching it off or something low) and do a simple running average outside the library where absolute values are used for the smoothing?

thanks in advance

olkal, 2022-11-03:
Hi!
Unfortunately it's not possible to filter out negative values, actually all values inside the data set are positive, they eventually become negative only after applying the tare/zero offset value that is subtracted from the final calculated average. 
I advice that you turn the smoothing of as you suggest, and do the smoothing in your own sketch.

Abasz, 2022-11-03:
Clear! thanks

closed 2022-11-03 by Abasz