olkal/HX711_ADC #115: Slow data readings

ilpantera, 2023-04-23:
Hi, first of all: thanks for the great work on this. I have an issue with the library though. While the readings are significantly better than with the other HX711 library, I have a la in data display. It takes about 2 seconds to get to a weight reading and once the weight is removed it takes another 2 seconds to get back to zero. I have this challenge both with the example code, as well as with my own adaption of it.
Any guidance how to speed this up?

One more question: In the Readme it is stated:

> Hardware and ADC noise: Wires between HX711 and load cell should be twisted and kept as short as possible.

Does this mean the dout and sck should be twisted and the gnd and vcc (in twisted pairs)?

Thanks!

olkal, 2023-04-24:
Hi!
There is no benefit to twist the dout and sck wires unless they are very long (these are digital signal wires).
The settling time with default settings and the HX711 in LOW RATE is 1.8 seconds. This is a result of the HX711 sample rate (HIGH or LOW) and the settings you have in the config.h file.  See description in the config.h file:

_Allowed values for "SAMPLES" is 1, 2, 4, 8, 16, 32, 64 or 128.
Higher value = improved filtering/smoothing of returned value, but longer setteling time and increased memory usage
Lower value = visa versa
The settling time can be calculated as follows:
Settling time = SAMPLES + IGN_HIGH_SAMPLE + IGN_LOW_SAMPLE / SPS
Example on calculating settling time using the values SAMPLES = 16, IGN_HIGH_SAMPLE = 1, IGN_LOW_SAMPLE = 1, and HX711 sample rate set to 10SPS:
(16+1+1)/10 = 1.8 seconds settling time.
Note that you can also overide (reducing) the number of samples in use at any time with the function: setSamplesInUse(samples)._

ilpantera, 2023-04-24:
Thank you so much for the response! What would be your recommendation with regards to the config.h in order to optimize for Accuracy & Speed? Is there a sweet spot for this or is it the 1.8sec?
Which wires would you twist between the load cell and HX711?

olkal, 2023-04-24:
There is no general sweet spot really, it depends on your requirements and it will always be a compromice between speed vs accuracy.

ilpantera, 2023-04-24:
Ok, thank you very much!

closed 2023-04-24 by ilpantera