olkal/HX711_ADC #116: UNITTTTTTTT!!!!!!!!!!!!!

McLover1790, 2023-06-24:
Can you provide in what units should our input wieght should be???
UNIIIITTTTTTTTTTTTT!!!!!!!!!!!!!!!!!!

olkal, 2023-06-24:
The library does not use any particular unit. The unit you choose for calibration known mass input will also be your output unit.

hotrock3, 2023-06-27:
@olkal , I think they are asking what measurement unit is being used. Is the code expecting grams? ounces?
I'm pretty sure it is grams, but I don't think I've seen it stated anywhere in particular.

olkal, 2023-06-27:
As I replied above, the library does not care or expect any particular unit: _**The unit you choose for calibration known mass input will also be your output unit**_
For example, lets say you use a known mass of 1000 grams when you do the calibration procedure (ref [Calibration.ino](https://github.com/olkal/HX711_ADC/blob/master/examples/Calibration/Calibration.ino) example sketch): If you now give the input 1(kg) when asked for known mass value, your output units will be kg. If you give the input 1000(gram) your output units will be gram. And so on...

hotrock3, 2023-06-27:
Ah, thank you for the clarification.

closed 2023-06-29 by olkal