olkal/HX711_ADC #117: Some error While test

Sudupatil, 2023-08-11:
Hello,

It's working fine except few exceptions.  Please go through the config.h and ESP32 code of my project. 

[config.h.txt](https://github.com/olkal/HX711_ADC/files/12319065/config.h.txt)
[Universal_Strength_24.txt](https://github.com/olkal/HX711_ADC/files/12319069/Universal_Strength_24.txt)

ESP32 Pin Configuration
DT  = 36
SCK = 33

Here is some line code of your interest to check

LINE 1139 While running sub
	Initializing_Scale()

Line 2277 When Click on Start Test

Now Observation are

1. When we start new test then it's showing last result. Where as i reset the variable =0.
2. Can i increase sampling speed. currently it's #define SAMPLES 32.
3. For same weight its showing fluctuated reading. but while doing calibration it's shows perfect every time.
	In my project there are 4 Modes. and when we select other mode then Direct this this happens. 

Project Use ILI9488 TFT display. 

Thanks and regards
Sudarshan Patil

Sudupatil, 2023-08-14:
I have done 
- initialize again before test begin.
- Reduced  setSamplesaInUse(4) from 32.

This fix the issues I am having

Thanks
Sudarshan Patil

closed 2023-08-14 by Sudupatil

olkal, 2023-08-14:
Thanks for the update.