olkal/HX711_ADC #118: No real issue - pls provide an example on how to use New funtions from v1.2.12

heinerolli, 2023-08-18:
As stated above - can you pls provide an example on how to use the new functions introduced in  v1.2.12
-> How do I use : dataWaitingAsync() and updateAsync()
I tried to replace (clumsily) 
     LoadCell.update() with LoadCell.dataWaitingAsync()
and 
     LoadCell.getdata() with LoadCell.updateAsync()
But that did not yield meaningful data

Thanks for your help!
Why I want to try this:
I get sometimes unexplainable flyer data from the HX711 - i.e. '200' - even though the 'real' range after calibration should only go up to '100'
I get more flyer data, when I add a 2nd HX711
That is why I want to try async access - by pure desperation...
(I am running them on 80Hz as I need the data fast (rather than with big accuracy)
I am running also an I2C dvice on seperate pins (INA 297) - can there be some sort of timing confligt??

In any case - thanks for your Library!!!

olkal, 2023-08-18:
Hi!
It's basecally just the update() function that is splitted into 2 new functions, one to check if new data is available and one to read the 24 bits of data and put the vaue into the dataset. 
So normally you would use: `if (LoadCell.update()) { i = LoadCell.getData(); }`
This can be replaced with: `if (LoadCell.dataWaitingAsync() { LoadCell.updateAsync(); i = LoadCell.getData(); }`
These functions was contributed in a pull-request, I'm not quite sure why anyone would need to use them. 

If you get wrong data sometimes it could be that you when this happen read the HX711 during the exact same time as it is updating it's internal register, this is an known issue with the HX711 and much more likely to happen at 80Hz rate and/or if you have other time-consuming code. I suggest that you try to use interrupt driven code for the HX711, see example code in the library (note that dout must be connected to an interrupt capable pin). If this don't solve it, you may try to temporarily disconnecting the i2c device or disable other code to isolate.

heinerolli, 2023-08-31:
Thanks for the feedback, @olkal 
Just a quick reply with my experience:
- I think, I am putting qouite some timing sensitive load on the Arduino, which adds to the problem with flyer data generation..
- Therefor using the interrupt based approach did not work, since it messed up other portions of the program (data tranmission via "SerialTransfer" library)
- Using the async approach did not make a difference
- -> So, I was not able to not get highflyxer data. 
- But using your 'flyer elimination' in config.h eliminated almost all the flyers (at the cost of samppling speed, but one can not have everything...):
`#define SAMPLES 					2
#define IGN_HIGH_SAMPLE 			1		//default value: 1
#define IGN_LOW_SAMPLE 			1`

So - thanks again!

closed 2023-10-16 by github-actions[bot]