olkal/HX711_ADC #119: Read_2x_load_cell example not working - potential issue with calling .begin() on multiple HX711_ADC objects

adamwhats, 2023-10-01:
Hi there,

I'm having some issues trying to run the [2x load cell example](https://github.com/olkal/HX711_ADC/blob/master/examples/Read_2x_load_cell/Read_2x_load_cell.ino) where I only get readings from LoadCell_1. I'm pretty confident there's no hardware issues as I can run the 1x example fine on each of my load cells independently.

As I've been messing around with it I've found that if I comment out the `LoadCell_1.begin()` call (alongside a few other lines to prevent errors) then suddenly I start to get readings from LoadCell_2. Interestingly it's not just a matter of which one I call `.begin()` on first as simply swapping the order round still results in only LoadCell_1 getting readings.

Another thing I've noticed is that the issue has some dependence on which pins are used for LoadCell_1 and LoadCell_2. When I use define the pins as so:
```
//pins:
const int HX711_dout_1 = 7; //mcu > HX711 no 1 dout pin
const int HX711_sck_1 = 6; //mcu > HX711 no 1 sck pin
const int HX711_dout_2 = 5; //mcu > HX711 no 2 dout pin
const int HX711_sck_2 = 4; //mcu > HX711 no 2 sck pin
```
then only LoadCell_1 works, but when I swap them round as below:
```
//pins:
const int HX711_dout_1 = 5; //mcu > HX711 no 1 dout pin
const int HX711_sck_1 = 4; //mcu > HX711 no 1 sck pin
const int HX711_dout_2 = 7; //mcu > HX711 no 2 dout pin
const int HX711_sck_2 = 6; //mcu > HX711 no 2 sck pin
```
then only LoadCell_2 works. It appears that the higher number pins seem to be taking preference?

I'd really appreciate any support, thank you,

Adam

olkal, 2023-10-01:
Hmm, that makes no sense, are you using the original example file without any additional code? You don't have any i2c/wire or other code mixed in that can access pin 4 and 5?

adamwhats, 2023-10-02:
Thanks for getting back to me. 

I put everything back on a breadboard and it worked fine! Turns out a bit of solder on the board I was using for the HX711's had connected the SCK from one with the DO from the other :sweat_smile: That explains why they worked separately but not together.

closed 2023-10-02 by adamwhats