olkal/HX711_ADC #120: help wanted: How to set Sampling rate from ESP32

Sudupatil, 2023-10-25:
Hello,

We can change sampling rate in library file, can we do that from ESP32 programing, instead change in library file.

Thanks  and regards

olkal, 2023-10-26:
Hi!
The sample rate 10SPS or 80SPS is a hardware (jumper) only setting on the HX711 module, so I guess you mean setting number of samples in moving average dataset `#define SAMPLES` in file config.h?
If so the answer is no. However, what you can do is to make a local copy of the library files and store them in your sketch folder.

Sudupatil, 2023-10-27:
Thank you for the clarification. In my project i would like to give user to
change runtime with menu.

Anyway I will set 8n config file and upload.

Thanks and regards
Sudarshan Patil

olkal, 2023-10-27:
When thinking about it just I recalled that I actually did made some functions that could be useful. It's just some time since I worked with this library so I had forgotten about it:

`void setSamplesInUse(int samples);   //overide number of samples in use`
`int getSamplesInUse();   //returns current number of samples in use`

Be aware that you have to compile the code with a dataset that is at least as big as you are going to call from the code/user input. E.g. if you set `#define SAMPLES 32` in config.h your max dataset size is 32. Then calling `setSamplesInUse(16)` will reduce the dataset from 32 to 16. However, `setSamplesInUse(48)` will not work.

Sudupatil, 2023-10-27:
Hello Komal,

Thank your very much. I will try it. And default will set higher (128). So
doesn't create any problem.

Thanks and best regards

Sudupatil, 2023-11-27:
I haven't got chance to test it yet. Perhaps it will work bcoz i haven't
get compile error.

Please close this issue. If there is any problem, i will post it.

Thanks & Regards

Sudupatil, 2023-11-27:
Have also created library for HX710B?

olkal, 2023-11-27:
No, I don't have a library for the HX710.

closed 2023-11-27 by olkal