olkal/HX711_ADC #125: tare at startup

ilker-aktuna, 2024-01-21:
I have a weird issue.

If at startup, I do tare like:
`LoadCell.start(stabilizingtime, true);`

it works fine. But it does tare when the device reboots (due to electricity outage)
I don't want to do tare at startup.
So I change it to:
`LoadCell.start(stabilizingtime, false);`

But when I do not do tare, it  does not measure correct after a reboot.

I read the saved calibration value from eprom and set it as calibration value at startup:

startup code:

```
 EEPROM.get(calVal_eepromAdress, calval);
  if(!(calval>0)) {
    calval=1.0;
  }
LoadCell.start(stabilizingtime, _tare);
  if (LoadCell.getTareTimeoutFlag() || LoadCell.getSignalTimeoutFlag()) {
    Serial.println("Timeout, check MCU>HX711 wiring and pin designations");
    while (1);
  }
  else {
    LoadCell.setCalFactor(calval);
  }
  while (!LoadCell.update());
  LoadCell.setCalFactor(calval);
```

what is wrong with my code ?

olkal, 2024-01-22:
Hi!
You need to save the zero-offset (tare) value and restore it at reboot.  See the example `Persistent_zero_offset.ino`.

ilker-aktuna, 2024-01-22:
ok. I didn't know that.
Do you have a documentation of this ?
Examples are fine but from their name sometimes what it does is not understood.

olkal, 2024-01-24:
No, unfortunately not. But I have now added a line in the readme file for this particular issue.

ilker-aktuna, 2024-01-24:
thanks. that's better than nothing :)
btw, I have another question:
when there is a change of load, the getData() method does not provide the final measurement at one shot.
I have to call it several times to get a good measurement.
So when there is a demand for measurement, I implemented like this:

                    for(int x=0;x<30;x++) {
                      boolean _resume = false;
                      while (_resume == false) {
                      if (LoadCell.update()) {
                        i = LoadCell.getData();
                        _resume = true;
                      }
                      }
                    }
                   finalWeight=i;

is this correct implementation?

closed 2024-03-10 by github-actions[bot]