olkal/HX711_ADC #128: How , do I set calibration factor for 2 different channels

thapliyal128, 2024-03-28:
Hey , so i have 2 load cells that I have connected to the A and the B channel , now in my code I want to get the readings for both the channels . 
Also , how can I set calibration value for the two sensors separately ?
Thank you !!1

olkal, 2024-03-28:
Hi!
Using both A and B channel is not really supported in this library, I strongly recommend to use 1 pcs HX711 for each load cell and A channel only. 

If you still want to try, it might work if you change the both the gain and the calibration factor (possibly also the zero-offset) every time you switch between the channels, but I never tested if this actually works. Obvious downsides by using channel B is that you can not make A and B channel conversions simultaniuosly so the overall speed and performance will suffer, and the channel B gain is only 32 (channel A max gain is128) so the general performance of channel B is not as good as for channel A.

thapliyal128, 2024-03-28:
So , for ex , if i want to set the calib factor for channel A , then i would have to switch to channel A gain and then specify , 
and then Switch to channel B to specify calib for channel B , 
Or , every time I switch channels , to get the values from diff load cells , i would have to set the calib and offset accordingly ?

Thank you

olkal, 2024-04-28:
Hi, sorry, I forgot to reply to you. But yes, you would have to set cailration and offset every time you switch.

closed 2024-04-28 by olkal