olkal/HX711_ADC #129: Compatibility issue with ESP32-S3 Dev Board

solrac3f, 2024-04-12:
Dear olkal,

I am writing to express my sincere appreciation for your incredible work on the HX711_ADC.h library. For quite some time now, I have been using this library with the ESP32 Dev Board, and it has consistently delivered exceptional results.

However, recently I encountered a compatibility issue when attempting to use the library with the ESP32-S3 Dev Board.  I have been unable to use the library with ESP32-S3 Dev Board.

I look forward to any updates or insights you may have regarding the compatibility of the HX711_ADC.h library with the ESP32-S3 Dev Board.

Thanks

olkal, 2024-04-12:
Hi, thanks!
What pins are you using for sck and dout on the ESP32-S3? 
Have you tried to change the value of  `SCK_DELAY` to 1 in file config.h?

solrac3f, 2024-04-12:
Hi olkal, 
Changing the SCK_DELAY to 1 in file config.h makes it work. I remember reading about it long time ago, only i forgot.
Appreciate your fast reply.

Thanks

olkal, 2024-04-12:
Great!

closed 2024-04-12 by olkal