olkal/HX711_ADC #131: How to read the value in setup method

Eashubh128, 2024-04-28:
Hey i have a query , if we can read the value in the setup method or not 
i have used the LoadCell.update() method before using the getData() . But its not working .
Please help , the weight i get in setup method is in negative and wrong , but when i do it in loop method it works . Please help

olkal, 2024-04-28:
Hi
The library is not really suited for single readings. 
You could try the following: run `LoadCell.update()` inside a `while()` loop until `getDataSetStatus()` returns `true`. Then call `getData()`. If you need it to complete faster, reduce the value of `SAMPLES` in file config.h.

Eashubh128, 2024-05-03:
Hey , i tried this ,  I have two loadcells connected , out of which  getDataSetStatus returns true for only 1 . 
please help

olkal, 2024-05-03:
Could you post your code, at least the `setup()` part?

closed 2024-06-17 by github-actions[bot]