olkal/HX711_ADC #136: How can you get more than one calibration factor?

chaba34, 2024-11-15:
Hi!
I'm a bloody beginner with arduino and programming, so I'm really interested in recommendations in order to learn new stuff.
Thank you so much for providing this library and the examples!
I'd like to measure weight using a setup with an uno R3, the hx711 and a load cell (range:0-50kg)
Unfortunatly, my project requires precise and on long term reliable values.
The problem is that using only one value with a given mass could result in poor measurement performance.
So I'd like to use at least three weights to calibrate the cell. 

Now, how can this be implemented in the calibration code? 
Also, should I, after getting those calibration-values, implement those in the "measure - code" using switch case* or is it better to plot those values and use a linear approximation in order to use an average calibration factor for the complete range of the cell?

*E.g. obtained 3 cal.- values for 10kg, 25kg and 45kg.  Then in the code use for the range 0-10kg value 1, for 10-25kg value 2 and value 3 for the rest of the range..

olkal, 2024-11-15:
hi

Calibration code: easiest is to use the code more or less as is, but repeating the process n times to get the calibration factors at each calibration point.

In the main sketch I would get the data from HX711 using cal factor 1.0. Then based on the data value decide which calibration range the data falls in to (range 1, 2, 3 or 4). Interpolate the calibration factors e.g. for 0-10 kg use value1, for 10-25kg use (value1+value2)/2, for 25-45kg (value2+value3)/2, for 45kg+ use value 3, and then divide the data value with the interpolated calibration value.

chaba34, 2024-11-15:
Hi! Thanks for your fast response!
That sounds like a very good idea!
Do you think implenenting the calibration values using switch case in the code(like the Read_1x_load_cell.ino), would be a good idea?

olkal, 2024-11-15:
Hi!
Using switch case vs. if/else is often a matter of taste, switch case is easy to read and fine for deciding an action from multiple exact integer values, but if/else is more flexible and probably what you want for this. For example, for the code below is it's better to use if/else: 

```
if ((myData >= 0) && (myData <= 1000)) {
  // do something
} 
else if ((myData > 1000) && (myData <= 2000)) {
  // do something else
}
```

closed 2024-12-31 by github-actions[bot]